astrochart/CHART #3: Organize project

adampbeardsley, 2018-07-19:
We should determine a structure for where files go in the repository. I would suggest something like this:
- /chart/
  - Our python code - turn into installable package so you can do things like `import chart`
- /gr/
  - Any gnu radio software we develop (e.g. if we write our own blocks)
- /tutorials/
  - Documents for tutorials. These could be broken into individual directories, each containing instructions and a flowgraph, and even a topblock if hacking is involved (e.g. the scanning spectrum)
- /Website/
  - content for the website

We can play with this to figure out what makes the most sense.

bryanna111, 2019-06-20:
I organized to the best of my ability, but the files (excluding 'read me' and 'index.html') that are on the top level don't have an option to be moved, just deleted. I'm not really sure what those files are so I'm going to leave them for now.

adampbeardsley, 2019-06-24:
I deleted and old config file and some meta_sink_orange thing that I don't understand. We can always recover them if needed, but I don't think we do.
I also moved photos of the team members into the website directory.
There is a file `CHART_LOGO_DRAFT_1.ai` which, based on the rest of that commit, looks like it was accidentally added. Did you intend to keep that file on the git repo?

bryanna111, 2019-06-24:
yeah, I added it to my local computer and it resultantly was added to the github repo. I'll delete that right now.

bryanna111, 2019-07-17:
I think organization is done

closed 2019-07-17 by bryanna111