astrochart/CHART #4: Logo!

adampbeardsley, 2018-07-19:
We need a logo for the website!
And a graphic for the top banner.

richardbmx, 2018-07-20:
@adampbeardsley  Hello my name is Richard I am a graphic designer and I love making contributions for open source projects, I see that you do not have a defined logo for your application. If you want I can do it for you without any problem you just have to authorize me to do it ... regards. ..

richardbmx, 2018-07-22:
@adampbeardsley  Happy day in this opportunity I present my proposal of a logo and icon for your project, the design is inspired by a radio antenna with a modern and minimalist design which makes it a unique and original design for your project

[image]

richardbmx, 2018-07-28:
@adampbeardsley  Hi, if you want any change in the logo, I'll gladly do it

adampbeardsley, 2018-07-28:
Hi Richard
Thanks for your interest. This is actually a student project, so I'm leaving it to the students to come up with their own logo.

Jaimedelavega, 2018-10-02:
@adampbeardsley  I have a CHART logo proposal! 

A little bit about my personal interpretation for the design elements below, but feel free to interpret in any creative way possible!

- The squares form a grid that allude to a chart, the namesake acronym of the project. Furthermore, the blocks coming together represent the hackable nature of the project - no single configuration of blocks is correct, assemble as you wish.

- The stars represent the application of the project, a telescope.

- The sine wave represents...  a radio wave!

Cheers!

[image]

[image]

adampbeardsley, 2018-10-02:
These look great @Jaimedelavega! I really like the concept, and you hit all the major components of the project. Could we iterate on a couple versions for different contexts? For example, it would be nice to have a "banner" version for the front page of the website: https://adampbeardsley.github.io/CHART/
I'd also like to have a very minimal one for icons (like a browser tab). For that I was thinking maybe just a square with a star in it. Or maybe incorporate a "C" somehow.

dannyjacobs, 2018-10-02:
Its beautiful Jaime! 
The squares, both re are also a callback to the Modern movement 
Another possibility for negative space, like the grid lines, but a little less subtle, would be a compass rose, which implies more overtly "cartography".

adampbeardsley, 2018-12-21:
Bumping this @Jaimedelavega - any chance you'll have time for iterating?

Jaimedelavega, 2018-12-22:
@adampbeardsley sorry for the delayed response. I have designed a handful of icons. Let me know which one (if any) fits the bill!

[image]
 
I also designed a banner for the website, but it is just the previously proposed logo surrounded by a sea of itself. I am open to suggestions and other ideas for a banner.  Also, what dimensions should the banner be for the best fit?
[image]

adampbeardsley, 2018-12-22:
Nice work Jaime!
I like 1, 3, and 6. 4 and 8 remind me of the Turkish flag. What do you
think @dannyjacobs?

I think the banner is perfect. I can try to put it on the website, and I’ll
let you know if it doesn’t fit. I’m not very experienced at making things
work for a variety of devices, so I’m sure it’ll work better for some than
others. But that’s fine at this stage.

Start thinking about picking out an old timey map after break. Happy
holidays!

adampbeardsley, 2019-02-12:
@Jaimedelavega can you be in the lab on Friday at 11am or 1pm? Need to wrap this up.

Jaimedelavega, 2019-02-13:
@adampbeardsley  I can be there around 1pm!

closed 2019-02-15 by Jaimedelavega