astrochart/CHART #14: Make "Download GNURadio and Collect your First Data" consistent

adampbeardsley, 2018-12-21:
The first tutorial has you use the QT GUI Sink block, but tells you to pick a filename, which it doesn't actually do.
Also look into soapysdr-module-all... it won't install, and I'm not sure it ever did.

bryanna111, 2019-06-20:
I think I resolved this; Could you look at my tutorial on google drive and let me know if that tutorial is what you were wanting.

adampbeardsley, 2019-06-24:
Great! It's definitely more consistent than it used to be. I went through and made a few comments to hopefully make it more clear/expand on it. There are a couple more things I'd like to make sure we do:
1) Address my comments in the google drive
2) "Play-test" the tutorial from scratch (start with a fresh raspberry pi)
3) Post the tutorial on the website.

These aren't really related to the original issue, so I'm going to close this one and make a new one.

closed 2019-06-24 by adampbeardsley