astrochart/CHART #48: Raspberry pi 4

adampbeardsley, 2019-06-24:
The new raspberry pi is out:
https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

I don't think we want to jump into it right away, but at some point we should have a look at it.

bryanna111, 2019-07-17:
Would you maybe want to buy one and check it out? It looks like it should run with our program and have faster processing speeds.

adampbeardsley, 2019-07-17:
Yeah, I tried to buy one, but everything's backordered. Then I read some people are having trouble with the USB-C connectors. So maybe we can wait a little until they're actually available and some bugs have been worked out.

adampbeardsley, 2019-08-22:
I went ahead and purchased a RPi4 (https://www.amazon.com/gp/product/B07VC9LWPL)
I'm going to use this issue as a log as I attempt to get it up and running.

First thing to note - Ubuntu Mate doesn't yet have a supported version for this pi. I'm going to try to use the experimental Raspberry Pi version, which is for ARMv8 (which is what the RPi4 has).
Current version is 18.04.2 (Bionic)

Plug it in... and nothing. Upon more googling, looks like Ubuntu Mate definitely does not work on RPi4 yet:
https://ubuntu-mate.community/t/support-for-raspberry-pi-4/19757/22

I'm going to try a basic raspbian install... ya never know.

adampbeardsley, 2019-08-22:
It looks like it worked! I'm able to do the tutorials up to and including "TAKE YOUR FIRST RADIO DATA," just with the adjustment of using raspbian instead of ubuntu mate. Hurray!
I'm going to switch over to looking at GPS stuff for a little while. But we should test this all the way through saving data with `collect.py`.

adampbeardsley, 2020-05-28:
Would be good to pick this up, and update the installation documentation where necessary.

adampbeardsley, 2022-06-06:
We've been using rpi4 for a while now. All that really needs to be done is update the installation documentation. I'm going to close this issue and open one on the website repo.