astrochart/CHART #146: Make portable

adampbeardsley, 2020-05-28:
Collapsible horn, box up electronics, etc.
model horn for different wavelength?

lmberkhout, 2020-06-04:
https://www.pulseelectronics.com/antenna_basic_concepts/
https://www.electronics-notes.com/articles/antennas-propagation/horn-antenna/theory-equations.php

adampbeardsley, 2021-11-02:
Not exactly what this issue is about, but we did make a portable system for the pi at WSU.
I bought this: https://www.amazon.com/dp/B07S74MP36
Then powered the pi with a [portable charger](https://www.amazon.com/Anker-PowerCore-Ultra-Compact-High-Speed-Technology/dp/B0194WDVHI), and powered the display with the pi.
Also got this little foldable keyboard: https://www.amazon.com/dp/B09DCKJRC8
overall it worked really well. We were able to do a few scans and only used maybe 1/3rd the battery.

The pi mounts right on the back of the display, so the plan is that each student can develop on their own pi, but when they're ready to take it out and record data, they just swap the SD card into the pi that's mounted.

[image]