astrochart/CHART #179: Workshop TODO

adampbeardsley, 2022-07-08:
- [ ] Wrap up data taking improvements - goal July 18
- [ ] Example analysis notebook
- [ ] fill in slides with lessons
- [ ] Come up with specific activities to do (with and without the telescope). APB needs to match these with Science Standards
- [ ] Choose a workshop dinner venue, make a reservation

adampbeardsley, 2022-07-18:
Can we see a CFL lightbulb?

dannyjacobs, 2022-07-18:
ASU Packing list
 - Fieldfox
 - low band bicone and LMR400 
 - blue antenna
 - power meter  (Adam has a laptop)

dannyjacobs, 2022-07-20:
> Any software I should install on laptop ahead of time?

https://www.minicircuits.com/softwaredownload/pm.html

adampbeardsley, 2022-08-19:
Never got the power meter working, but there's a separate issue for that (#175) and the workshop is over, so closing this.

closed 2022-08-19 by adampbeardsley