astrochart/CHART #203: RTL-SDR Blog v4 dongle support

adampbeardsley, 2024-05-20:
The new RTL-SDR.com blog dongle is out, but they have not pushed the drivers to osmosdr yet. I'm going to try and get it working, and document my progress here.

EphemeralDisco, 2024-05-20:
Instructions for v4 driver installation are here: https://www.rtl-sdr.com/v4/

For *nix:
The first set of instructions for Linux (Debian) did not work for me. The second set of instructions using packages did work. Copied below:
Update RTL-SDR driver to v4:
sudo apt update
sudo apt install libusb-1.0-0-dev git cmake
sudo apt install debhelper

git clone https://github.com/rtlsdrblog/rtl-sdr-blog
cd rtl-sdr-blog
sudo dpkg-buildpackage -b --no-sign
cd ..

sudo dpkg -i librtlsdr0_*.deb
sudo dpkg -i librtlsdr-dev_*.deb
sudo dpkg -i rtl-sdr_*.deb

adampbeardsley, 2024-05-20:
I was able to get it to work using the first set of instructions on https://www.rtl-sdr.com/v4/

I tested things using a simple gnu radio companion flowgraph:
[image]

Here was my test data using v3 dongle, tuned to a local radio station (seen very weakly at 89.5 MHz), plus the RF explorer transmitting at 89.7 MHz (unsure why it's showing up at 89.6 MHz).
[image]

Before installing the new driver, the flowgraph happily ran using the v4 dongle, but produced no signal:
[image]
(note I forgot to set the "Display RF Frequencies" button, but it's tuned the same as the first dataset.)

After going through the instructions on rtl-sdr.com/v4, and relaunching GRC, I got a missing block error. 
I reran this line:
`sudo apt install gnuradio-dev gr-osmosdr librtlsdr-dev`

And now everything seems to work. New data test:
[image]

And to be sure, I ran the chart data collection with the RF explorer transmitting at 1419.7 MHz:
[image]
The bias-T was successfully enabled, and we get a nice bright line at the correct frequency. This was done in the lab, so there's a ton of other cruff, unsurprisingly.

PR #217 (2024-06-22): update Install and notebook touchup
supercedes #207 , 
Closes #204 
Closes #203
closed 2024-06-22 by adampbeardsley