astrochart/CHART #204: CHART_GUI in python virtual environment (VENV)

EphemeralDisco, 2024-05-20:
Unable to get CHART_GUI working in python virtual environment.

per adafruit:
Starting with the October 10, 2023 [Bookworm release](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/) of the [Raspberry Pi OS](https://www.raspberrypi.com/software/), the use of [Python Virtual Environments](https://docs.python.org/3/tutorial/venv.html) (venv) when pip installing packages is required. No more sudo pip. This will break things and require learning new things. Yeah.

lmberkhout, 2024-05-20:
Hello! Could you please provide a little more detail on this issue so we can help solve it? What is the error you are running into trying to get CHART_GUI working in the venv?

adampbeardsley, 2024-05-21:
I believe the issue is that our gnu radio install is global (via the sudo apt install commands), so the virtual environment doesn't see them. Here's the workaround that seems to work for me:

First install gnuradio etc. (note if you're using the rtl-sdr blog v4 dongle, you'll need to install that driver according to https://www.rtl-sdr.com/v4/ (issue #203 ))
`$ sudo apt install gnuradio-dev gr-osmosdr librtlsdr-dev build-essential git cmake`

Next, create a virtual environment that can see the global installs:
`$ python -m venv --system-site-packages ~/myenv`
Activate the venv:
`$ source activate myenv/bin/activate`

Now continue on with the python installs:
```
(myenv) $ pip install ipython
(myenv) $ pip install numpy
(myenv) $ git clone https://github.com/astrochart/CHART.git
(myenv) $ cd CHART
(myenv) $ pip install .
```

At this point, it appears everything is properly installed. The only issue is that the setup script quietly failed to create the GUI link on the desktop... not sure why. But that's ok, you can simply run it from the terminal (with your venv activated):
`(myenv) $ gui.py`
That should launch the data collection GUI.

I'll make a PR with these instructions in the README. @EphemeralDisco Let us know if this fixes your issue.

EphemeralDisco, 2024-05-21:
This seems to do the trick!  Thank you for helping us figure this out!

Just FYI - The default RPi Bookworm image does not include xterm which is
required by GNU radio, so I'm adding this to our instructions:
sudo apt-get install xterm

Also, for activating the venv, this command from above did not work: $
source activate myenv/bin/activate
but it did work without "activate" before the path:  $ source
myenv/bin/activate

chris

*Chris Huebner* (he/they)
Design and Innovation Specialist
St. Paul Academy and Summit School <https://www.spa.edu/>

adampbeardsley, 2024-05-21:
Sorry about the typo! Glad this worked... let us know how the observing goes!

PR #217 (2024-06-22): update Install and notebook touchup
supercedes #207 , 
Closes #204 
Closes #203
closed 2024-06-22 by adampbeardsley