astrochart/CHART #205: Module Not Found Error / Pip Install Won't Work

saraj1228, 2024-05-21:
[image]
[image]

Have everything set up for CHART, including GNU Radio, but having issues opening the CHART package regarding customtkinter. Have tried to pip install but am struggling, I'm relatively new to coding so any help would be appreciated!!

saraj1228, 2024-05-21:
never

adampbeardsley, 2024-05-21:
Hi @saraj1228 , glad you got it to work, but could you tell us what you did? That way if someone else runs into the issue they can see your solution. Thanks!
(By the way, I think this is related to issue #204 that I'm working on right now)