astrochart/CHART #209: Bias-T Instructions

lmberkhout, 2024-06-05:
Be more clear about what the Bias-T is and when it is used, so your students don't forget to turn the amplifiers on!

adampbeardsley, 2024-07-09:
Is this meant to go on the website? Or are you asking for clarification in the GUI?

lmberkhout, 2024-07-09:
Both would be good

lmberkhout, 2024-07-09:
redundancy is always nice

Alexandra202421, 2024-07-11:
I'm not seeing anywhere that Bias T is mentioned on the website but I know there used to be something there. Should I add a new section to the instructions for it assuming I'm not missing it somewhere?

adampbeardsley, 2024-07-11:
I noticed this yesterday... apparently we lost a whole section of the tutorial that explains how to use the GUI when we did the website update. Whoops. How git-ambitious are you feeling to go back in time and recover the old tutorial? If not, I can probably take a crack at it tomorrow.

Maybe this is related to the issue where danny asked for a GUI image from the new disk image.

Alexandra202421, 2024-07-11:
I can certainly give it my best go but it might take me a while lol, I'll see what I can do!

Alexandra202421, 2024-07-11:
I found the datataking section in merged pull request 50 and its definitely gone by danny's "clean up sidebar" request but I can't find where it disappeared in the middle of that, not sure if these images will post either

[image]
[image]

PR #232 (2024-07-30): bias t change
closes #209
closed 2024-07-30 by adampbeardsley