astrochart/CHART #225: Remove colons from filenames

adampbeardsley, 2024-07-17:
Windows does not like colons in file names. This was causing windows to give up on cloning.
I made a band-aid by renaming the testdata in #224 , but we need to update more globally to not make files with colons.
Also need to update analysis tutorial to point to updated testdata dirs

Alexandra202421, 2024-07-17:
on the topic of windows related issues, is there any way to make the chart.iso install faster? The analysis code requires the package to be installed but it takes about 3 hours to install on windows. I'm also having some other issues running the code too. the first line errored as "ModuleNotFoundError: No module named 'ipympl'" which led me down a rabbit hole trying to enable path access for anaconda and failing, reinstalling anaconda, getting an error about ipykernel not installed, struggling to install it, and then having to also reinstall matplotlib since I started over with anaconda. is this excessive just to get the analysis code to run?

adampbeardsley, 2024-07-22:
closed in #226 
windows installation issue solved over slack.

closed 2024-07-22 by adampbeardsley